Ryochan7/sc-controller #6: State of the fork

C0rn3j, 2020-07-13:
First of all thanks for forking and maintaining this.

What is the current status of the fork, is there something missing even?

sc-controller is the last thing I have on Python 2 which I'd love to finally get rid of.

Ryochan7, 2020-07-13:
I will attempt to keep this fork active for a while. The current plan is to keep this fork as a soft fork only focusing on Python 3 compatibility while merging changes from the upstream Kozec repo. I really don't want to start making backwards incompatible or feature changes that would require hard forking the project. Maybe it will be desired at some point though.

I am not sure if any functionality is still missing from this fork. I have only been modifying routines as I have run into problems making mappings. Most of the main features and action types work fine now. One area that might be lacking is support for controllers other than the Steam Controller. It does work for the DS4 v.1 as well but I am not so sure about support for other controllers.

Right now, I have a pre-made AppImage hosted on my Google Drive. It is confirmed to work with Ubuntu 20.04, Fedora 32, and Xubuntu 19.10. I did not even want to make an extra tag in this repo so that a proper Release could be hosted on GitHub. I think it would be safe to dub this fork v0.4.8 and not interfere with any version number that will be used in the upstream repo. I probably should get around to doing that.

SC_Controller-x86_64_20200707.AppImage
https://drive.google.com/file/d/16QHJ4d5eaCYCYPdmFQ96DWJLh81L7gX8/view?usp=sharing

This project is my main foray back into gaming more on Linux. I have a decent Linux game (and VN) backlog and it would be nice to actually play them through on Linux.

C0rn3j, 2020-07-15:
Could you please tag a release like upstream does https://github.com/kozec/sc-controller/releases ?

I'd like to convince the maintainer of AUR/sc-controller to use this fork but I'd be having a hard time without a release.

Ryochan7, 2020-08-03:
Finally got around to putting a release out there. I also uploaded an updated AppImage for the release.

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8
https://github.com/Ryochan7/sc-controller/releases/download/v0.4.8/sc_controller-0.4.8.glibc-x86_64.AppImage

leycec, 2020-08-14:
**Brilliant.** Since you're doing all the heavy lifting here, @Ryochan7, *and* since @kozec has publicly disowned Python 3 (and thus this fork), it might be advisable to either remove the Liberapay and Paypal links redirecting to @kozec *or* revise them to reference your own donation-ready accounts (assuming you have any, of course). [Patreon](https://www.patreon.com/) might be a preferable option.

Honestly, @kozec's C-based rewrite strikes me as a catastrophic misstep. C is well-regarded as a ludicrously low-level non-portable systems language, but... as a high-level portable responsive desktop GUI apps language? Not so much. You might well find your fork rapidly becoming the de-facto drop-in replacement for upstream across major Debian- and RedHat-based distros as well as the plethora of hardcore rolling-release distros (e.g., Arch, Gentoo).

In short: *strap in.* If you ever wanted a full-time open-source gig, you now have the ticket. :tickets: :rocket:

Ryochan7, 2020-08-15:
I am not so sure about that. For now, I don't plan on deviating too far from the upstream repository apart from changes needed for Python 3. Any funding for SC Controller should still goes towards Kozec to help with the C version.

C0rn3j, 2020-10-26:
Closing as my question was answered and releases have too been a thing for a while.

As a side note, AUR/sc-controller now uses this fork.

closed 2020-10-26 by C0rn3j