Ryochan7/sc-controller #7: OSD menu severely broken

C0rn3j, 2020-07-15:
Am talking about this menu
[image]

Turn Controller OFF
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/scc/sccdaemon.py", line 942, in press
    menuaction.button_press(mapper)
  File "/usr/lib/python3.8/site-packages/scc/special_actions.py", line 255, in button_press
    self.execute(mapper)
  File "/usr/lib/python3.8/site-packages/scc/actions.py", line 550, in execute
    return self.execute_named(self.SA, mapper, *a)
  File "/usr/lib/python3.8/site-packages/scc/actions.py", line 545, in execute_named
    return getattr(sa, h_name)(mapper, self, *a)
  File "/usr/lib/python3.8/site-packages/scc/sccdaemon.py", line 319, in on_sa_osd
    self._osd('message', '-t', action.timeout, '-s', action.size, action.text)
  File "/usr/lib/python3.8/site-packages/scc/sccdaemon.py", line 299, in _osd
    data = b"OSD: %s\n" % (shjoin(data) ,)
  File "/usr/lib/python3.8/site-packages/scc/tools.py", line 126, in shjoin
    s = [ x.encode("utf-8") for x in lst ]
  File "/usr/lib/python3.8/site-packages/scc/tools.py", line 126, in <listcomp>
    s = [ x.encode("utf-8") for x in lst ]
AttributeError: 'int' object has no attribute 'encode'
```

Autoswitch Options
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/scc/osd/menu.py", line 553, in on_event
    self._selected.callback(self, self.daemon, self.controller, self._selected)
  File "/usr/lib/python3.8/site-packages/scc/osd/menu.py", line 482, in show_submenu
    self._submenu.parse_argumets(["menu.py",
  File "/usr/lib/python3.8/site-packages/scc/osd/menu.py", line 216, in parse_argumets
    items = self.items.generate(self)
  File "/usr/lib/python3.8/site-packages/scc/menu_data.py", line 29, in generate
    items.extend(i.generate(menuhandler))
  File "/usr/lib/python3.8/site-packages/scc/x11/autoswitcher.py", line 341, in generate
    win = X.get_current_window(menuhandler.xdisplay)
  File "/usr/lib/python3.8/site-packages/scc/lib/xwrappers.py", line 352, in get_current_window
    trash, prop = get_window_prop(dpy,
  File "/usr/lib/python3.8/site-packages/scc/lib/xwrappers.py", line 334, in get_window_prop
    prop_atom = intern_atom(dpy, prop_name, False)
ctypes.ArgumentError: argument 2: <class 'TypeError'>: wrong type
```

Display Keyboard
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.8/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.8/site-packages/scc/sccdaemon.py", line 668, in handle
    instance._sshandler(self.connection, self.rfile, self.wfile)
  File "/usr/lib/python3.8/site-packages/scc/sccdaemon.py", line 732, in _sshandler
    self._handle_message(client, line.strip(b"\n"))
  File "/usr/lib/python3.8/site-packages/scc/sccdaemon.py", line 852, in _handle_message
    if not self._can_lock_action(client.mapper, SCCDaemon.source_to_constant(l)):
  File "/usr/lib/python3.8/site-packages/scc/sccdaemon.py", line 1029, in _can_lock_action
    if what in SCButtons:
  File "/usr/lib/python3.8/enum.py", line 310, in __contains__
    raise TypeError(
TypeError: unsupported operand type(s) for 'in': 'str' and 'EnumMeta'
```

Ryochan7, 2020-07-17:
I almost always play wired so the Turn Controller OFF feature was not tested. Only some more recent private custom edits makes mouse emulation behave like I expect when the controller is connected to the dongle. That problem should be easy to fix.

Display Keyboard works fine on my system. I have fixed that same type of error in other parts of the code so it should be easy to fix that problem.

I have never used the Autoswitch system. The problem with the xlib wrapper would likely impact some other portions of the application as well.

Ryochan7, 2020-07-22:
All of these issues have been fixed in commit e3d2b78b03fb21280a2e39b0c3304d05ca82ff8b. It didn't take long to fix the problems once I got around to working on it. Currently my autism has kept me busy experimenting with the Switch Pro and Joy-Con controllers.

C0rn3j, 2020-07-22:
Just got my Valve Index today so I'll be busy too, but I tested it and it works. I'll switch AUR/sc-controller-git today and try convincint the non-git maintainer to do the same

closed 2020-07-22 by C0rn3j