Ryochan7/sc-controller #10: Working Controllers

toby63, 2020-08-04:
As I saw this:
`One area that might be lacking is support for controllers other than the Steam Controller. It does work for the DS4 v.1 as well but I am not so sure about support for other controllers.`
Comment: https://github.com/Ryochan7/sc-controller/issues/6#issuecomment-657757132

I thought it might be a good idea to report working and non-working controllers.

toby63, 2020-08-13:
8BitDo SN30 Pro is working.

Ryochan7, 2020-08-15:
I just added an SDL Game Controller mapping for my DS4 v.1 and now it fully works in SC Controller. The mapping was a bit off when trying to use the Controller Registration dialog window to make a custom mapping.

https://github.com/Ryochan7/sc-controller/commit/e80e2ead6c052818c197fb26ceaa0fe94670510c

Ryochan7, 2020-08-15:
I guess this leads to another query. Should I sync up the gamecontrollerdb.txt file in this repo with the version from the upstream repository?

https://github.com/gabomdq/SDL_GameControllerDB

toby63, 2020-08-15:
> I guess this leads to another query. Should I sync up the gamecontrollerdb.txt file in this repo with the version from the upstream repository? Will only copy over the Linux platform mapping strings.
> 
> https://github.com/gabomdq/SDL_GameControllerDB

Sounds interesting.
Looks like the file in kozecs repo was last updated 3 years ago.
But maybe there a differences between the files?

Ryochan7, 2020-09-14:
Got a new version release with an updated gamecontrollerdb.txt mapping file. There are a lot more Linux mappings available so hopefully it helps out more people. There were also several portions of code that were updated based on reports from pyflakes.

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.1

Bleuzen, 2020-12-24:
@Ryochan7 The PS5 (DualSense) controller seems to be not working. Is there support planned?
Btw. kernel patches are up now, maybe this helps figuring out how the controller works:
https://patchwork.kernel.org/project/linux-input/list/?series=404369

Ryochan7, 2021-01-02:
The only way that support for the DualSense would be added would be when an entry is added to the gamecontrollerdb file. I wouldn't be surprised if the upstream file already has a mapping for it. I have not done any gaming on Linux since getting my DualSense. DS4Windows has been keeping me busy especially with adding support for more non-DS4 controllers.

I will check the latest SDL GameControllerDB mapping file later today.

S1ngl3-x, 2022-05-07:
I have heard DS4Windows now supports dualsense. Can we hope to see this in foreseeable future too?

Ryochan7, 2022-05-07:
Probably not. In theory it might be possible to add support for the DualSense into the current DS4 support module. The DualSense input report format is not too different from the DS4 input report format. My priority has been to deal with other problems within SC Controller. I have not used my DualSense in a long time and it is definitely my least used controller now.

S1ngl3-x, 2022-05-11:
Dualsense now works as confirmed in this Reddit thread https://www.reddit.com/r/linux_gaming/comments/uleupu/dualsense_linuxgang_heads_up_0487_of_sccontroller/i83himn/?context=3 . Tested myself.

lranixonl, 2022-10-27:
> The only way that support for the DualSense would be added would be when an entry is added to the gamecontrollerdb file. I wouldn't be surprised if the upstream file already has a mapping for it. I [...]

Dualsense is now in the GameControllerDB, search as PS5 Controller

[](https://github.com/gabomdq/SDL_GameControllerDB/blob/master/gamecontrollerdb.txt)

Ryochan7, 2022-10-27:
That's cool. Probably should pull the latest GameControllerDB anyway. Although, I was modifying the original work on the ds5drv module and got Bluetooth support working with hidraw rather than evdev. Got stuck on converting the IMU data to a quaternion. Also, been busy working on DS4Windows stuff. Need to get back to experimenting with DualSense support and possibly finish it up.

https://github.com/Ryochan7/sc-controller/tree/theY4Kman-dualsense

lranixonl, 2022-10-27:
No problem, if you return to work on it I can be a beta tester. I need it to you with Lutris and AC3 remastered.