Ryochan7/sc-controller #11: Can't edit gyro already set, can't apply new gyro bindings

Supreeeme, 2020-09-24:
When I try to edit a gyro binding that I've already set, I get this error in terminal:    

	Traceback (most recent call last):
	  File "/usr/lib/python3.8/site-packages/scc/gui/controller_widget.py", line 268, in on_click
	    self.app.show_editor(self.id)
	  File "/usr/lib/python3.8/site-packages/scc/gui/app.py", line 385, in show_editor
	    ae.set_input(id, action)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 1129, in set_input
	    self.set_action(action)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 895, in set_action
	    self.enable_preview(self._action)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 955, in enable_preview
	    enabled = action.strip().get_previewable()
	  File "/usr/lib/python3.8/site-packages/scc/modifiers.py", line 884, in strip
	    return self.mods.values()[0].strip()
	TypeError: 'odict_values' object is not subscriptable

This error appears even when I only open and close the gyro binding without setting anything, and then try to go back to set something. I have to right click and clear to be able to open it again.

And then when I try to add a new gyro binding, I get this error after setting my binding:

	Traceback (most recent call last):
	  File "/usr/lib/python3.8/site-packages/scc/gui/ae/first_page.py", line 120, in on_lblMarkup_activate_link
	    self.editor.on_link(link)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 228, in on_link
	    self.set_action(action, from_custom=True)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 895, in set_action
	    self.enable_preview(self._action)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 955, in enable_preview
	    enabled = action.strip().get_previewable()
	  File "/usr/lib/python3.8/site-packages/scc/modifiers.py", line 884, in strip
	    return self.mods.values()[0].strip()
	TypeError: 'odict_values' object is not subscriptable

And then this error when I try to press OK to set it (but nothing happens, as you can imagine)

	Traceback (most recent call last):
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 488, in on_btOK_clicked
	    a = self.generate_modifiers(self._action, self._selected_component.NAME=="custom")
	AttributeError: 'NoneType' object has no attribute 'NAME'

Ryochan7, 2020-09-24:
I'll have to take a look at this later today. I almost never utilize the Steam Controller gyro so this problem would not have impacted me.

Ryochan7, 2020-09-26:
This issue seems to be fixed with commit 23aa9a3ccccacde1aefd85618c98747657f2c1b7.

Supreeeme, 2020-09-27:
Looks great, thanks

closed 2020-09-27 by Supreeeme