Ryochan7/sc-controller #12: Setting gyro to either stick disables pads, errors with right stick

Supreeeme, 2020-09-27:
More gyro issues lol.
So if I set the gyro to the right stick, it appears to be set fine, but these pop up in the terminal:

	W ActionEditor  selected_component no longer handles edited action
	W ActionEditor  <scc.gui.ae.gyro_action.GyroActionComponent object at 0x7f488edfc670>
	W ActionEditor  mode(RPADTOUCH, gyro(Axes.ABS_RY, Axes.ABS_RX), None)
	W ActionEditor  (mode(RPADTOUCH, gyro(Axes.ABS_RY, Axes.ABS_RX), None))

And if I try to edit it after setting it to right stick, these errors pop up:

	Traceback (most recent call last):
	  File "/usr/lib/python3.8/site-packages/scc/gui/controller_widget.py", line 269, in on_click
	    self.app.show_editor(self.id)
	  File "/usr/lib/python3.8/site-packages/scc/gui/app.py", line 385, in show_editor
	    ae.set_input(id, action)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 1129, in set_input
	    self.set_action(action)
	  File "/usr/lib/python3.8/site-packages/scc/gui/action_editor.py", line 901, in set_action
	    if component.handles(self._mode, ActionEditor.strip_modifiers(action)):
	  File "/usr/lib/python3.8/site-packages/scc/gui/ae/gyro.py", line 82, in handles
	    action = action.mods.values()[0]
	TypeError: 'odict_values' object is not subscriptable

I also notice that the pads and the gyro no longer do anything. All of the buttons seem to be fine though.
A similar thing happens if I try to set gyro to the left stick, but none of these errors pop up and I can edit it fine. But the pads and gyro are still disabled.

Ryochan7, 2020-10-02:
Thank you for spotting this issue. Debugging this problem also led to a few other changes as well. The Gyro Enable/Disable Button option was busted in the mapper. Once the trigger was first activated then the mapper would break. That issue was fixed in commit 8b71c721399c2ecc04db71f89574b519208650ac.

Ryochan7, 2020-11-29:
Finally got around to making a new version and AppImage file

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.2

closed 2020-11-29 by Ryochan7