Ryochan7/sc-controller #14: Steam Controller randomly disconnects when playing the Witcher 3

Araly, 2020-11-01:
I'm using the controller with the USB dongle.

I'm playing the Witcher 3, everything goes fine until a fight occurs, and the controller dies, shuts down.

Then I press the steam button, it turns on again, and the light flashes, the controller is usable again,  I can move the cursor around and so. It stays in this state for a while.

If I return to the game, I can expect the controller to randomly shut off again at any point. I've been trying to see if it correlates with me pressing some specific key but it doesn't seem like it.

I don't have many games out of steam, so the Witcher 3 is kind of the only game I use sc-controller on at the moment, but I'll add info if I play another game and see other issues.

The gui told me it was recently updated to v0.4.8.1, so maybe that's a hint.

Let me know if there are other information I can give.

C0rn3j, 2020-11-15:
This sounds more like an issue with upstream sc-controller rather than the Python 3 fork, do you have the same issue with upstream sc-controller?

Ryochan7, 2020-11-20:
I never liked how vibration works with SC-Controller so I have the option disabled. Unfortunately, I do not have a way to test that scenario. My compy's GPU does not support Vulkan so I have no way to use Proton to run the Witcher 3 on Linux.

Araly, 2020-11-25:
Not long after this, the batteries died. Maybe the rumbling was too much for the low batteries ? Anyways, I changed the batteries and now this is not an issue anymore, so I'll close

closed 2020-11-25 by Araly