Ryochan7/sc-controller #16: [Feature Request] Enable OSD on Wayland compositors using wlroots

trigg, 2020-11-16:
While Wayland itself is missing a central overlay/osd system the compositors that are based on wlroots have layer shell which allows this feature.

I've cobbled together a proof-of-concept fork and if there's interest I can work more on it. Tested on sway and wayfire.

[Fork here](https://github.com/trigg/sc-controller/tree/wlroots)

Main issues:
- Ugly (code) implementation
- All OSD is pinned to bottom left, a configurable option would help.

Ryochan7, 2020-11-20:
I will have to give this a try sometime later. It has been a while since I have tested Wayland in any capacity.

trigg, 2020-11-20:
Thank you :) This is the final thing I need to be completely on Wayland on my gaming setup.

Ryochan7, 2020-11-23:
I tried this in Fedora 33 running Gnome on Wayland. The overlay does not work for me.

trigg, 2020-11-23:
Gnome is not a wlroots compositor.

https://way-cooler.org/book/wlroots_introduction.html

The only exceptions I've seen for wlroots are Gnome, Plasma and Weston. Everything else should be wlroots

Ryochan7, 2020-11-23:
I have tried it in Sway and the OSD still does not display for me. SC-Controller does not lock the input as it should either.

trigg, 2020-11-23:
So any error messages? What's said about Wayland in the log?

Random swing in the dark but the usual first step is ensure you've installed `gtk-layer-shell`

It will only lock if it manages to enable an OSD, thankfully.

Ryochan7, 2020-11-23:
DNF shows that the package is already installed. SC-Controller does print a message that Wayland has been detected.

```
Wayland detected. Disabling X11 support, some functionality will be unavailable
```

When the Steam button is pressed and the daemon is running with debug output.

```
Cannot show OSD; there is no scc-osd-daemon registered
```

Ryochan7, 2020-11-23:
Damn it. Didn't realize that the code was in a different branch. Going to have to build the program again and try it there. Sorry.

trigg, 2020-11-23:
Ah my apologies. I thought I had linked the branch too!

Ryochan7, 2020-11-23:
Okay. Now it mostly works. Some behavior issues currently are that submenus do not display (All Profiles) and the keyboard will not hide when B is pressed.

trigg, 2020-11-23:
Having 'ESC' does not work but set it to 'hide keyboard' works

[image]

trigg, 2020-11-23:
Also on intel gpu it renders the submenu in the same spot but seems to accept input just fine. Ideally we'd offset from the edge of screen

trigg, 2020-11-23:
[image]
[image]

Ryochan7, 2020-11-29:
I ended up merging your current changes in with commit e026fc17e1c782ba903feeab28b421db130fd89d. Hopefully these changes will help out other people who primarily use Wayland. I will continue to stick with Xorg. A new build will probably be made public in a few days. Thank you for the changes.

Ryochan7, 2020-11-29:
Version 0.4.8.2 is now out and has these changes incorporated. I will keep this issue open for a while in case you want to work on this further.

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.2

meZeex, 2022-11-15:
Using sway 1.7 and sc-controller 0.4.8.9
The OSD menus (inclyding keyboard) aren't working

@trigg are you still using this and is it working for you or do you know how to hack this in a working state?

meZeex, 2022-11-15:
thanks :)

yeah, with steam my issue is, that there is no menu support for desktop configuration, which is my main goal as i'd like to use it as a keyboard

maybe i can help?
i mainly come from js/node.js development, but if things are documented well enough, i should have a good chance

(some time ago i wanted to try to just do a touchpad keyboard, but evdev and the likes didn't give me enough documentation...)

meZeex, 2022-12-04:
ok, i dunno how, menus are working now (radial doesn't)
virtual keyboard doesn't work

sadly for me, the menus don't work the way i hoped for and navigation is really bad

aqxa1, 2023-02-10:
Using sway/wlroots this implementation partially works, but the menu is rendered behind fullscreen windows (so it can't be seen with games/fullscreen video players etc).

TimB87, 2024-01-21:
For me, OSD doesn't work with a current wlroots/sway setup.