Ryochan7/sc-controller #17: scc set-profile doesn't work due to socket being opened in read-only mode

aqxa1, 2020-11-24:
Default for makefile() in Python 3 looks to be read-only mode, so set-profile fails to work:
```Traceback` (most recent call last):
  File "./scc", line 9, in <module>
    main()
  File "/usr/lib/python3.7/site-packages/scc/scripts.py", line 389, in main
    sys.exit(command(sys.argv[0], sys.argv[2:]))
  File "/usr/lib/python3.7/site-packages/scc/scripts.py", line 136, in cmd_set_profile
    print("Profile: %s" % (profile,), file=s)
io.UnsupportedOperation: not writable```

Setting https://github.com/Ryochan7/sc-controller/blob/39690b388b618794dd04f091669e81c6d42086b6/scc/scripts.py#L286 to:
s.makefile(mode="rw") fixes the problem. I'm not sure if there are any side effects to this, but it works as a workaround at least.

Ryochan7, 2020-11-28:
I have never run into this problem but I can try out that change. I don't think there should be any side effects.

Ryochan7, 2020-11-29:
The code change seems to work fine for me. The change has been added in commit f8deac4e43b2a1f0331c79aecc298b1bbc645bed.

Ryochan7, 2020-11-29:
Made a new AppImage file and made a new release. Thank you for bringing up this problem.

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.2

closed 2020-11-29 by Ryochan7