Ryochan7/sc-controller #18: [Notice] Version Conflict

Ryochan7, 2020-11-28:
Kozec has pushed a version 0.4.8 build of SC-Controller. That messes up the original assumption that using a 0.4.8.x range would be considered safe. 0.4.7.x should have been used for releases of the old Python 2 port and 0.4.9.x was for the beta C port. That is no longer valid.

https://github.com/kozec/sc-controller/releases/tag/v0.4.8

There is supposed to be hip fire trigger support in the latest Python 2 build but I don't know if I will end up incorporating it in the Python 3 code. Will need to examine the diff more closely. This whole issue might finally justify a true hard fork.

Ryochan7, 2021-11-16:
Almost a year since this was posted. Feels like a time skip happened.

It looks like the upstream project is mostly dead. There has been no activity in any branches since 2021/03/03 and it looks like Kozec has not done any activities on GitHub in months. Also, I will likely not maintain this fork of the project much longer. At some point, I would like to make some sort of mapping project to replace my use of SC Controller at least for my own use. The main goal of this project was to use the Steam Controller to catch up on my Linux game backlog, specifically Deus Ex: Mankind Divided, and I mostly have; some games don't even work for me anymore.

Although my current experiments have been on the Windows side using C# / .NET, I would like to use Qt to work on a new mapper for use on Linux. It has been an idea that I have had for years but still haven't accomplished. My main problem currently is that I cannot design a way to replicate the Action Layer system used in Steam Input. I keep running into faults when I try to design something. I have a working prototype with Action Set support.

Current prototype:
https://gitlab.com/ryochan7/switchpromappertest/-/tree/action_ideas

Supreeeme, 2021-11-17:
I'd be willing to maintain sc-controller if you lose interest.
Is that new mapper idea a universal controller mapper? It sounds interesting. Hope it has SC support :)
What's the issue with action layers if you don't mind me asking? It seems like it would just be similar to a mode shift except you shift multiple buttons at once.

Ryochan7, 2021-11-20:
Mainly been trying to duplicate a lot of Steam Input functionality with the prototype. Duplicating the activator system has been a problem especially when it comes to switching an Action within an Action Layer. I have been using the Stopwatch class to keep track of elapsed time but that approach doesn't work when trying to switch Activators (ActionFunc in the code iirc) in an Action Layer. There has to be some form of independent state that gets transferred between instances during an Action Layer switch. Although the current prototype only works with the Switch Pro controller, I would want something that would support some of the other controllers I have.

After all these years, Steam Input still does not interpolate Xbox 360 axes properly and I keep verifying it over time. Steam Input still uses an output range of [-32767, 32767] for an Xbox 360 stick axis. That is wrong and the one missing unit in the negative axis direction matters for some games; the correct range is [-32768, 32767]. My goto examples for games affected by the lack of one unit are Bioshock 2 and Duke Nukem Forever. There is no way to perform a full speed camera turn from the left. Since Steam Input is proprietary, there is no hope that it will ever be fixed and it looks like Valve doesn't care.

Ryochan7, 2022-04-07:
Finally made some progress on my prototype. Been actively coding over the past month. Came up with an API for Action Layer stacking and got many more action classes created. Just in the last few days, I ported over the current mapper progress to my SteamControllerTest project and I now have a configurable mapper for the Steam Controller. The syntax of the JSON files will likely change as I might just get rid of the InputMappings abstraction from the profile action objects.

It definitely feels better to me than what Steam Input or reWASD offers. There is still a lot of work to do regarding the Touchpad and Trigger action types. Most future mapping work will likely still be done in the switchpromappertest project. Not too worried about having a real GUI still. The current app needs a JSON profile file passed as a cmd argument or the program will exit.

Current test branch:
https://gitlab.com/ryochan7/steam-controller-test/-/tree/action_ideas

Ryochan7, 2022-04-17:
If anyone is curious, here is a demo video showing the current progress of the test mapper. Mostly shows XInput bindings with Mouse-like Joystick bound to Right Pad on Action Layer 1. Action Layer 2 is used to bind Right Pad to a normal absolute Xbox Right Stick for some game segments. An overlay shows up a few times in the video when layer switching occurs.

Remember Me - Steam Controller mapper test
https://www.youtube.com/watch?v=-QvJoQMGupg

Not even sure how far this version will go. I do not know if I would get around to porting the project to C++ / Qt.

Ryochan7, 2022-05-04:
Running out of motivation. Just a small bit of progress. More output action types for Triggers, Touchpad, and Gyro controls. Multiple Steam Controller support although it has only been tested with one. Changing profiles in the app from a combobox. Still no profile editor so editing is still done directly in the JSON. Temporary device specific options window for Touchpad coordinate rotation so those are not hard coded in the mapper anymore; I have been using -18 degrees for Left Touchpad and 8 degrees for Right Touchpad.

Probably nothing else worth mentioning. I know I have some other tasks I need to get done in this repo.

[image]