Ryochan7/sc-controller #21: DataFrog's SNES-like controller needs updated gamecontrollerdb entry

jose1711, 2020-12-30:
With the default `gamecontrollerdb.txt` file `sc-controller` is not detecting DPAD on Data Frog's SNES-like controller (https://www.playconsoler.com/data-frog-2-pack-retro-snes-classic-pc-game-controller-windows-10-gamepad-joysticks-usb-game-controller-mac-linux-joypad-joystick-for-raspberry-pi-3-steam-super-nintendo/). The following change is required:

```diff
--- /usr/share/scc/gamecontrollerdb.txt.orig	2020-12-17 20:26:02.295016903 +0100
+++ /usr/share/scc/gamecontrollerdb.txt	2020-12-17 20:33:03.152176951 +0100
@@ -248,7 +248,7 @@
 03000000321500000009000011010000,Razer Serval,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a5,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a4,rightx:a2,righty:a3,start:b7,x:b2,y:b3,platform:Linux,
 050000003215000000090000163a0000,Razer Serval,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a5,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a4,rightx:a2,righty:a3,start:b7,x:b2,y:b3,platform:Linux,
 0300000032150000030a000001010000,Razer Wildcat,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux,
-03000000790000001100000010010000,Retrolink SNES Controller,a:b2,b:b1,back:b8,dpdown:+a1,dpleft:-a0,dpright:+a0,dpup:-a1,leftshoulder:b4,rightshoulder:b5,start:b9,x:b3,y:b0,platform:Linux,
+03000000790000001100000010010000,Data Frog SNES controller,a:b2,b:b1,x:b3,y:b0,back:b8,start:b9,leftshoulder:b4,rightshoulder:b5,leftx:a0,lefty:a1,platform:Linux,
 0300000081170000990a000001010000,Retronic Adapter,a:b0,leftx:a0,lefty:a1,platform:Linux,
 0300000000f000000300000000010000,RetroPad,a:b1,b:b5,back:b2,leftshoulder:b6,leftx:a0,lefty:a1,rightshoulder:b7,start:b3,x:b0,y:b4,platform:Linux,
 030000006b140000010d000011010000,Revolution Pro Controller,a:b1,b:b2,back:b8,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b12,leftshoulder:b4,leftstick:b10,lefttrigger:a3,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b11,righttrigger:a4,rightx:a2,righty:a5,start:b9,x:b0,y:b3,platform:Linux,
```

jose1711, 2020-12-30:
Related: https://github.com/kozec/sc-controller/issues/610

Ryochan7, 2021-01-02:
Will do

Ryochan7, 2021-02-26:
Almost 2 months later and the change has finally been made. Changed in commit 48c807cd4bbf6af7b7254eac466a3d8686f500b3. Been too busy

Ryochan7, 2021-03-18:
Version 0.4.8.4 has this change now

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.4

closed 2021-03-18 by Ryochan7