Ryochan7/sc-controller #22: Using two steam controllers at the same time?

FrankyB, 2021-01-02:
Dear Ryochan7, 
I discovered an issue lately regarding using two steam controllers simultaneously, and I found a similar topic on https://github.com/kozec/sc-controller/issues/609
When pluging in two steam controllers, the latest controller does not react as expected - it behaves so badly that for months I was thinking one of it was broken - until I was using the "broken" one solemnly and it worked without issues.
As you can read on the forementioned link, I was just the second to discover this issue. It is, as jose1711 stated, no difference using the controllers via USB dongle or directly via cable.

I checked out versions
v0.4.8.2 => not working
v0.4.8.1 => not working
v0.4.8 => not working
v0.4.7 => not running at all (expects python2)
v4.9.1 => not running at all (expects python2)
v4.9.10 => not running at all (expects python2)
 
Switching on the second controller gives no error message at all. Just a "W App           Daemon reported unknown profile: 'None'; Overriding." but the controller misbehaves badly (it summons the OSD pressing steam button, but unable to navigate in it and unable to close it)

Anything I can do helping in this issue?

Ryochan7, 2021-01-02:
I only have one Steam Controller so I cannot test the scenario directly. I believe I have tried to test using a Steam Controller and a DS4 in SC Controller at the same time but I could be mistaken. I can give that a try sometime later today.

Ryochan7, 2021-02-26:
Finally tried it last night. It looks like SC Controller only performs mapping for one controller. I don't have a way to test the old Python 2 code anymore so I have no way to test if some changes in this code broke functionality. Also, in the GUI, changing the profile for one controller changes the associated profile for every connected controller.

berarma, 2021-04-06:
I've tried to debug this issue but I have little knowledge about Python and the code.

The reason the OSD gets stuck seems to be that the daemon is locking the input from the other controller. But the input events from the other controller are ignored because the controller name doesn't match.

This could happen because the controller from the mapper object is the wrong one. There's one mapper per controller but maybe both point to the same controller device. This could happen while adding controllers or maybe when reusing created mappers.

I'll try to have another look when I have time but someone more familiar with the code might find it sooner.

commit 8d650e5a (2021-04-07): Using two Steam controllers. Fixes #22
closed 2021-04-07 by Ryochan7