Ryochan7/sc-controller #23: No module named 'scc'

mozzribo, 2021-01-17:
Hi.

I receive this when trying to run:
```
Traceback (most recent call last):
  File "/usr/bin/sc-controller", line 16, in <module>
    from scc.tools import init_logging
ModuleNotFoundError: No module named 'scc'
```

Any ideas on what might be causing the issue? Running from this build:
https://aur.archlinux.org/packages/sc-controller

On Manjaro KDE.

Alastor27, 2021-02-07:
Maybe you have Python2 installed as the default instead of Python3?
Make sure, the **/usr/bin/python** link pointing to the **python3** executable.

mozzribo, 2021-02-08:
All seems to be in order on that front! Anything I could do to run some tests?

Alastor27, 2021-02-09:
Well, maybe I'm wrong, but at the moment I still can't think anything else then somehow the script tries to run with the wrong python version.
You can check, which version you are using by typing in a shell `python -V`
Also you should check, if the **scc** python module is really installed in the system by typing in a python console `help("modules")`. If **scc** is in the list, it's installed.
Or the same using `pip` just execute `pip list | grep sccontroller`)

By the way, how did you installed the package? With an AUR package manager like trizen/yaourt or manually?

mozzribo, 2021-02-09:
There was no scc module. I de- and reinstalled the package (using Pamac) and now it works. Thanks for the help! Let me know if there's any other info that I can provide.

closed 2021-02-09 by mozzribo