Ryochan7/sc-controller #25: Service not stopping after quitting program

julikaiba, 2021-02-08:
I might be missing something but when I right click and click quit in the taskbar the icon vanishes and I cant find the program in the taskmanager. However the settings still persist and when pressing the steam button, even the little popup window shows to let me switch between profiles or open the keyboard etc. 
Is there another background service or daemon that needs to be stopped? (this became a problem after the autoclicker i was trying to build using the repeat function didnt stop anymore and even quitting the program didnt change that fact)

Ryochan7, 2021-02-26:
The default behavior is that the daemon keeps running after the GUI is closed. I don't like that default. I always change the setting on my end by enabling the **Automatically Disable Emulation When Closing GUI** option in the Settings window (located under the Advanced tab).

julikaiba, 2021-02-26:
Thats exactly what I was looking for, thank you very much.

closed 2021-02-26 by julikaiba