Ryochan7/sc-controller #27: sc-controller GUI can't connect to daemon after Bluetooth Steam controller is connected

ramesh45345, 2021-02-20:
I am not able to get sc-controller's **GUI** to recognize my Steam controller when paired using Bluetooth (the daemon does recognize it). I am using version `sc_controller-0.4.8.2.glibc-x86_64.AppImage`. Here are the steps I used:

1. Run `./sc_controller-0.4.8.2.glibc-x86_64.AppImage daemon debug`, and then `./sc_controller-0.4.8.2.glibc-x86_64.AppImage`.
2. Observe that a `Permission denied: '/dev/hidraw3'` error occurs (see first log below).
3. Run `chmod a+rw /dev/hidraw3` to make that device read/writeable by scc.
4. Close and reopen the daemon and GUI.
5. Observe the second error which seems to occur due to processing some message? (see second log below)

**Note**: Once I run that chmod command, and run the GUI, the GUI seems to try to access the daemon, which procues the traceback. After this point, the GUI causes this traceback in the daemon whether it is plugged in via the USB dongle or bluetooth. The traceback seems to occur in either case after that.

I understand the logs below probably don't contain enough information to figure out what is going on. For example, I'm not sure what string python was trying to parse that it failed on, or even why it has that error now and not beforehand. Please let me know if I can produce extra logs to figure out what is going on here.

My system is a currently updated Fedora 33 x86_64, which uses python 3.9 (not sure if thats relevant).

First run of scc daemon:
```
./sc_controller-0.4.8.2.glibc-x86_64.AppImage daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
W evdev         Failed to enable Evdev driver: 'python-evdev' package is missing.
D USB           Registered USB driver for 054c:09cc
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7f31688f4c10>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7f31688f4ca0>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7f3168994190>
D SCCDaemon     Created control socket /var/home/user/.config/scc/daemon.socket
D SCCDaemon     Connected to XServer :1
E SCBT          [Errno 13] Permission denied: '/dev/hidraw3'
Traceback (most recent call last):
  File "/tmp/.mount_sc_conRsjfMH/usr/lib64/python3.8/site-packages/scc/drivers/sc_by_bt.py", line 105, in new_device_callback
    dev = HIDRaw(open(os.path.join("/dev/", hidrawname), "w+b"))
PermissionError: [Errno 13] Permission denied: '/dev/hidraw3'
D USB           USB device added: 28de:1142
D OSD DaemonCtrl    Connected to daemon, version 0.4.8.2
D OSD DaemonCtrl    Daemon is ready.
I SCCDaemon     Registered scc-osd-daemon
I OSD osd.daemon    Sucessfully registered as scc-osd-daemon
```

Second run of scc daemon, after permission changes:
```
./sc_controller-0.4.8.2.glibc-x86_64.AppImage daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
W evdev         Failed to enable Evdev driver: 'python-evdev' package is missing.
D USB           Registered USB driver for 054c:09cc
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7f9784e61c10>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7f9784e61ca0>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7f9784f03190>
D SCCDaemon     Created control socket /var/home/user/.config/scc/daemon.socket
D SCCDaemon     Connected to XServer :1
D SCCDaemon     Assigned default_mapper to <SCByBt scbt1>
D SCCDaemon     Turning gyrosensor ON
D SCCDaemon     Controller added: <SCByBt scbt1>
D OSD DaemonCtrl    Connected to daemon, version 0.4.8.2
D OSD DaemonCtrl    Daemon reported profile change for scbt1: /var/home/user/.config/scc/profiles/retroarch.sccprofile
D OSD DaemonCtrl    Daemon is ready.
I SCCDaemon     Registered scc-osd-daemon
I OSD osd.daemon    Sucessfully registered as scc-osd-daemon
----------------------------------------
Exception occurred during processing of request from 
Traceback (most recent call last):
  File "/usr/lib64/python3.9/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib64/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib64/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/tmp/.mount_sc_conQTwNBZ/usr/lib64/python3.8/site-packages/scc/sccdaemon.py", line 682, in handle
    instance._sshandler(self.connection, self.rfile, self.wfile)
  File "/tmp/.mount_sc_conQTwNBZ/usr/lib64/python3.8/site-packages/scc/sccdaemon.py", line 746, in _sshandler
    self._handle_message(client, line.strip(b"\n"))
  File "/tmp/.mount_sc_conQTwNBZ/usr/lib64/python3.8/site-packages/scc/sccdaemon.py", line 834, in _handle_message
    to_observe = [ x for x in message.split(":", 1)[1].strip(" \t\r").split(" ") ]
TypeError: a bytes-like object is required, not 'str'
----------------------------------------
----------------------------------------
Exception occurred during processing of request from 
Traceback (most recent call last):
  File "/usr/lib64/python3.9/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib64/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib64/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/tmp/.mount_sc_conQTwNBZ/usr/lib64/python3.8/site-packages/scc/sccdaemon.py", line 682, in handle
    instance._sshandler(self.connection, self.rfile, self.wfile)
  File "/tmp/.mount_sc_conQTwNBZ/usr/lib64/python3.8/site-packages/scc/sccdaemon.py", line 746, in _sshandler
    self._handle_message(client, line.strip(b"\n"))
  File "/tmp/.mount_sc_conQTwNBZ/usr/lib64/python3.8/site-packages/scc/sccdaemon.py", line 834, in _handle_message
    to_observe = [ x for x in message.split(":", 1)[1].strip(" \t\r").split(" ") ]
TypeError: a bytes-like object is required, not 'str'
----------------------------------------
```

Ryochan7, 2021-02-26:
Fixed with commit 3ab9627cd6986a164ef3e1f4cc21cadf9111b201. A similar change was needed for a few other commands as well. Now I just need to get a new release out at some point.

Ryochan7, 2021-03-18:
Version 0.4.8.4 is now out.

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.4

closed 2021-03-18 by Ryochan7