Ryochan7/sc-controller #32: Poor right trackpad polling rate over Bluetooth

Supreeeme, 2021-04-24:
When using sc-controller over BLE, the polling rate of the right trackpad is significantly lower than over USB, and even lower than the polling rate of the left trackpad put into mouse mode. With `evhz`, I get an average polling rate of 110 Hz over USB and around 50 Hz over bluetooth. This also results in the trackpad feeling really laggy and bad.

Ryochan7, 2021-04-28:
I would usually either use a USB connection (almost always) or the dongle. Is the Steam Controller Bluetooth poll rate really that low? I would probably consider that low of a poll rate unusable.

Anyway, one issue is that the mouse routine does **not** take the gamepad poll rate into account so the mouse will end up being slower as the poll rate for the controller decreases. I have a diff that changes the mouse routine to take poll rate into account and adjust the distance to move the mouse cursor accordingly. That change is not in this repository because I didn't want to introduce another conflict with the main upstream Kozec repository; need to merge some of the more recent changes from upstream.

Supreeeme, 2021-04-28:
The bluetooth poll rate (without sc-controller) is actually about the same as over USB according to evhz, but just less consistent (for obvious reasons). It's only really low when sc-controller is controlling the controller.

Ryochan7, 2021-05-04:
Trying BT out now and I am getting poll rates around 10 - 20 ms mostly. Mouse control while connected via BT is definitely worse than when using a USB connection.

Supreeeme, 2021-05-06:
So it seems like this is caused by the haptics, specifically the frequency. If I turn the haptics off, the mouse is as responsive as over USB/Dongle. If I turn haptics on and lower the frequency, the lag gets worse.

Supreeeme, 2021-05-08:
So I've been looking into this over the last few days. I don't know if this is fixable? Outside of avoiding too low of a frequency over Bluetooth. It seems to be a result of more ioctl calls in a shorter amount of time with a higher frequency (via HIDRaw.sendFeatureReport). I can't think of why it would result in delayed inputs over Bluetooth vs USB/dongle outside of Bluetooth accepting the calls at a slower rate?

Ryochan7, 2021-06-19:
I never use haptics with SC Controller. It adds quite a bit of delay to the mapping thread. It looks like the HIDRaw.sendFeatureReport routine uses the Control endpoint so I am going to assume that is a blocking call. It should probably be in its own separate thread if that is the case.

My mouse settings are good enough that I don't need to the assistance of the haptics to get a sense of mouse speed. I might just end up adding some of my custom changes into the repository. It would likely change the feel of the mouse cursor overall.

Ryochan7, 2021-07-01:
Finally got around to making some commits. The mouse move routine has been changed to take gamepad poll time into account. A couple of other small changes have been added in recent commits that improve the feel of the virtual mouse. A couple of changes have been left out from my custom diff as they would change the feel of the mapper routines a bit too much (changing default sensitivities primarily). At least now, I can actually play with the Steam Controller wirelessly (using the dongle) and not feel that the cursor is throttled.

I really want to get around to playing some Deus Ex: Mankind Divided on Linux soon. Only ever booted up the Linux port once just to make sure it would run on my system.

Ryochan7, 2021-07-08:
Finally released a new build of SC Controller. The most recent mouse changes are available.

closed 2021-07-08 by Ryochan7