Ryochan7/sc-controller #37: Cannot import Steam Profiles

TheLux83, 2021-06-18:
Hey there,

I wanted to try to import my Steam Controller configs to SC-Controller, but I get the following error message:

```
local variable 'vdfpath' referenced before assignment
Traceback (most recent call last):
  File "/home/the_lux/git/sc-controller/scc/gui/importexport/import_vdf.py", line 375, in import_vdf
    self._profile.load(filename)
  File "/home/the_lux/git/sc-controller/scc/foreign/vdf.py", line 597, in load
    data = parse_vdf(open(filename, "r"))
  File "/home/the_lux/git/sc-controller/scc/lib/vdf.py", line 32, in parse_vdf
    if not vdfpath:
UnboundLocalError: local variable 'vdfpath' referenced before assignment

Profile filename: /home/the_lux/.local/share/Steam/userdata/10609250/241100/remote/2458639198_controller_config.vdf
```

I get this error, when I clone the repository and start `./run.sh`
When I use the AppImage, I don't get any error message at all.

Can someone help me?

Supreeeme, 2021-06-19:
This might be a dependency issue. Do you have [python-vdf](https://pypi.org/project/vdf/) installed? Forgot to add that to the dependencies. Also probably not checking for the package correctly.

If you're running Arch, there's an AUR package for it.

TheLux83, 2021-06-19:
Hey Supreeeme. It seems that this was the issue. I've installed `python-vdf` via `pip` with `pip install vdf` and now it works.

I'm using Manjaro, but when I install the AUR package, I had the same error message. So I needed to install python-vdf here as well.

But now it seems to be running well :-) 
Thanks for your help!

Ryochan7, 2021-06-19:
SC Controller comes with its own VDF file reader utils so python-vdf should not be necessary. I don't have that python package installed on my Fedora system. Granted, I have not tried the latest changes that fix importing Steam profiles yet.

Ryochan7, 2021-06-19:
The VDF reader utils works fine on my end with the latest fix. Installing python-vdf should not be necessary.

carloscheddar, 2021-07-22:
Fixed for me after installing `python-vdf`. I'm using Arch and installed `sc-controller` via `AUR` using `yay`.

Ryochan7, 2021-07-23:
I have to check the AppImage and see if the build script needs to be updated. The README was updated before the last release to list python-vdf as a dependency.

Ryochan7, 2021-07-23:
Importing Steam profiles does not work in the current AppImage. I will re-open this issue until that problem is fixed.

Ryochan7, 2021-07-23:
Found a couple more issues with VDF importing. Got them fixed in commit f67249c90adeb09563e620f3c66d7f0b180e6f40. I won't worry about closing this issue again until the next public release.

closed 2021-11-16 by Ryochan7

Ryochan7, 2021-11-30:
Should be fixed in commit 03ea18529e5f751af871f5d5b01c72f408b80361