Ryochan7/sc-controller #38: [Feature Request] Include DS4 v1 emulation

italoghost, 2021-08-01:
I used to use DS4Windows with DS4v1 emulation on my DS4 v2 to make some games recognize the DS4 prompts correctly. Would it be possible to add this same feature on sc-controller?

Ryochan7, 2021-08-01:
I am not really sure tbh. uinput is definitely capable of emulating a DS4 controller.

The python code seems like it is abstracted enough that it could create a virtual DS4 controller; it looks like virtual devices are likely created on daemon startup rather than creating them on demand. The profiles use the event keys for the Xbox 360 controller so some mapping abstraction would have to be added to convert those events to the proper values expected from a DS4 controller.

As it is, I am not sure if there is much of a need to emulate a DS4 controller in SC Controller.

italoghost, 2021-08-01:
@Ryochan7 I see.. Do you know a way in which I could have this functionality on Linux? I know that it is just a little annoyance, but seeing Xbox prompts on games that I know that have native DS4 prompts really bother me

Ryochan7, 2021-08-15:
I have not looked into it yet. However, after all the comments that have popped up with DS4Windows from users thinking their PC should magically become a PS4 with DS4 controller emulation enabled, I cannot trust most users to look into whether a game in question actually supports a DS4 before posting bug reports. Filing this under can't have nice things. Sorry.

closed 2021-08-15 by Ryochan7