Ryochan7/sc-controller #40: sc-controller not passing gyro/accel info to Yuzu

octoshrimpy, 2021-08-16:
Most of the body is here:
https://github.com/kozec/sc-controller/issues/647

basically, yuzu adds a "shake to detect" feature to enable gyro, but it looks like accel values are locked(?) at 0, so yuzu never detects the accel, therefore it never enables gyro.

Ryochan7, 2021-08-18:
I barely worry about the UDP server functionality when it comes to DS4Windows. I have never tried it in SC Controller. I will try to find out why the accel values stay stuck at 0.

Ryochan7, 2021-08-18:
Okay. None of the gyro assigned actions use the accel values. The raw gyro rotation values and gyro quaternion values are passed to the cemuhook special action. I am not sure the quaternion values can be used to approximate accel values. Not sure how I would try to patch accel values into that special action without breaking other gyro actions.

Ryochan7, 2021-08-18:
SC Controller never actually pulls the accel data from the controller into its own controller state data. The byte offsets that I am encountering are slightly different than I have used when testing things on Windows; byte offsets seem to be 1 lower. I think I might have found the proper offset for AccelX but I need to test in my Windows program to confirm the data.

https://gitlab.com/ryochan7/steam-controller-test/-/blob/master/SteamControllerTest/SteamControllerLibrary/SteamControllerReader.cs#L255

octoshrimpy, 2021-08-18:
I am on linux, and willing to test code as well.

Ryochan7, 2021-08-18:
Got the accel data pulled from the Steam Controller. Got an initial draft up for testing. The orientation is very likely wrong right now as I cannot remember what the CemuHook expects for accel axis directions. Too bad there is no PadTest like app for Linux to test data going through the UDP server.

Test branch:
https://github.com/Ryochan7/sc-controller/tree/cemuhook_test_changes

Ryochan7, 2021-08-18:
Fixed the accel coordinate mappings in commit ba38aa76f365e1f90d4dfe17de46c8260b25565c. PadTest.exe runs fine under Wine so that helped confirm that the changes fixed the orientation. Also had to use the code from steam-gyro-for-cemu for a reference.

octoshrimpy, 2021-08-19:
I've pulled that branch and followed installation instructions in readme, and then ran ./run.sh. 

Hopping in Yuzu, and attempting to map gyro still did not work. 
I'm not seeing any place to see logs or values out of sc-controller..

Ryochan7, 2021-08-20:
Did you make sure to run `git checkout cemuhook_test_changes` before running run.sh? I have accidentally not had git use the proper branch when cloning a repo for testing before. The default branch for the git repo is the python3 branch.

Also, which connection are you using? The changes would work for a Steam Controller connected via USB or the SC Dongle. No change has been made for a Steam Controller connected via Bluetooth yet.

kristianbrox, 2021-08-21:
I have been following @octoshrimpy's breadcrumbs and tried the test branch. I am still not able to map the motion controller in yuzu.
I am a rookie git user, so I might have missed something. However `git status` prints:
`Your branch is up to date with 'origin/cemuhook_test_changes'.`
I'm not sure if this is right, but to run sc-controller, I ran "build/scripts-3.9/sc-controller" 
In sc-controller I cleared all the gyro mappings, chose "CemuHook motion provider" as output mode, and "Always Active" under Gyro enable button. 
My steam controller is connected wirelessly with the steam USB dongle.

I am happy to test more.

athdesk, 2021-08-21:
I'm on the new branch also, as you can see, but still I'm getting no accelerometer data on yuzu. I'm going to check again in case I'm doing something wrong, and I'll test the controller via cable as opposed to wireless

[image]

Ryochan7, 2021-08-22:
Not sure if the kernel driver for the Steam Controller would play into this. The only change that I have made is that the configure method has been changed to set 0x1C for self._enable_gyros rather than 0x14. The 0x08 flag is supposed to be used for enabling the accelerometer but I was getting the data without it so I opted to not change the Python code; my .NET app uses 0x1C when configuring the Steam Controller. I am primarily testing this on a USB connection.

dbff3f66d5baef01c6c4656bc8a88306a33f786f

Also, I would recommend that people try out the official PadTest app from rajkosto. It works under Wine 6.13 on my Fedora box.

https://files.sshnuke.net/PadTest_1011.zip

kristianbrox, 2021-08-22:
I tested with usb cable, and padtest registered the accel-values, and it worked fine in yuzu. I also tested your latest patch, and now gyro+accel works wirelessly with the steam dongle as well. Thank you so much!

However, I discovered one other problem with yuzu. The controller turns off when it receives a vibration signal from yuzu (vibration can be disabled though). This seems to work fine with usb cable connected though.

Ryochan7, 2021-08-22:
Got the accel values from a Bluetooth connection. Rebuilding the sc_by_bt C library will be necessary. e3e9ab682abed735b6f6b85f3ab467ce1d8dd497

```
python3 setup.py build
```

The force feedback problem is likely an old bug in SC Controller regarding reading wrong feedback values from the virtual Xbox 360 controller. I remember it being a problem when playing Saints Row 4 a long time ago so I have always left rumble support disabled in SC Controller.

Ryochan7, 2021-09-01:
Ended up merging these changes to the python3 branch. Not sure when a new AppImage will be made with these changes.

Ryochan7, 2021-09-09:
Put out a new release with these changes

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.6

closed 2021-09-09 by Ryochan7