Ryochan7/sc-controller #42: Circular Mouse doesn't accept mousewheel direction settings

LinuxJoeUser, 2021-09-17:
Being able to set the direction of the circular wheel input could be a useful improvement, so that turning the wheel clockwise would act as if the wheel were being turned counter-clockwise.  

Currently, the natural place to do this is by selecting one of the "Output buttons" boxes after selecting "Circular" and "Wheel" would allow me to choose a wheel direction.  However, selecting a wheel direction doesn't enter the option like it does with everything else on the pop-up.  The only way to clear the pop-up is to choose something else or simply close the pop-up.

[image]

For example, I scroll using the *inside* edge of the pad, so that when I swipe up with my thumb (clockwise) I expect the content/list/menu to go up as well, like it does on my phones, tablet, and desktop.

Ryochan7, 2021-09-18:
I am still very hesitant to add features that would further deviate this repo from the upstream Kozec Python 2 repository. I have tested changing the invert options for Sensitivity and that seems to do nothing for the circular action type. If I would try to change the behavior of the circular action, that would probably be the place I would attempt to change.

Ryochan7, 2021-09-18:
Got it working okay with commit 826423a67c59a0304a8024f6af6e8eda0613ad50. The BallModifier class treats a child CircularModifier instance slightly differently than other child actions. The speed settings needed to be carried over to the circular modifier. Now, inverting the X axis allows the mouse wheel direction to be reversed and not change the behavior for different contained child actions. That seems to be good enough to get the job done.

LinuxJoeUser, 2021-09-21:
Thanks for getting a work-around in so quickly!  I'll close this.  I get your hesitancy around adding to the original code.  Could you give me a sense of what sorts of things I might report/request here, and I'll report everything else with Kozec.

Thanks for keeping SCC alive on Python.  After the Steam Controller itself, SCC is my favorite thing about the Steam Controller!  Cheers!