Ryochan7/sc-controller #43: DualShock 4 no longer detected properly when plugged in

gddgddg, 2021-10-23:
When I plug in my DualShock 4, in the past the software would detect it and automatically switch to it. This no longer happens; instead, nothing happens, but I can register it as a "Sony Computer Entertainment Wireless Controller". If I register it this way, it doesn't detect the gyro, which is important for certain types of games. The controller is detected by other software, such as Steam, in which I can use the gyro and touchpad just fine.

I've tried this on two machines both running Arch Linux, where the issue persists.

Ryochan7, 2021-11-16:
Not that sure. My DS4 v.2 is detected fine but my DS4 v.1 does not get detected.

Ryochan7, 2021-11-16:
Only DS4 v.2 support was available in the ds4drv driver in SC Controller. Made a quick change so a Sony DS4 v.1 could also be detected. That works now as well.

https://github.com/Ryochan7/sc-controller/tree/76ebfefa4d88564d82f033baaa0b53b236bc8f88

gddgddg, 2021-11-28:
Seems to be working fine now, thank you very much.

closed 2021-11-28 by gddgddg