Ryochan7/sc-controller #48: fix enable evdev support for non-steam controller on fedora 35

EtereosDawn, 2021-12-04:
python-evdev package not found on fedora 35, I cannot register my generic control. I also get other errors without accessing /dev/uinput. I have all dependencies installed including python-evdev/python3-evdev. but I think it's a version problem... is it possible to fix it?

Ryochan7, 2021-12-08:
I am still running Fedora 33. Haven't had a reason to update. I will take a look at the version of python-evdev available for Fedora 35 and see if there is some sort of regression.

The /dev/uinput issue is likely just a permission problem that requires some slight configuration. Adding a special udev rules file should take care of the problem.

https://github.com/Ryochan7/sc-controller/blob/python3/scripts/69-sc-controller.rules

Ryochan7, 2021-12-12:
Finally looked into this. Made a few commits to attempt to fix the Controller Registration dialog. evdev support is confirmed to work properly with the recent changes. Not so sure about raw USB HID support based on my tests with the DS4 with the special ds4drv module disabled. Maybe it would work fine with game controller devices.

There were a couple of type conflicts occurring with file reading changes in Python 3.

closed 2022-04-17 by Ryochan7