Ryochan7/sc-controller #49: Cannot create new profiles with CSDs disabled

tyzoid, 2021-12-30:
While trying to create a new profile, no create/save button appears when CSDs are disabled. Enabling CSD for this app reveals an 'OK' button in the title menu.

[image]

Additionally, pressing the Enter/Return key does not create a profile while the text cursor is in the textbox.

Ryochan7, 2022-04-17:
I cannot duplicate the problem with the OK button. I would assume there is no event set up to catch Enter being pressed when that textbox has focus. Not too worried about that one though.

Ryochan7, 2022-05-05:
Not much to work on here. I cannot duplicate the original problem so I am going to close this issue.

closed 2022-05-05 by Ryochan7