Ryochan7/sc-controller #52: Official-driver style "Joystick Camera" smooth

mochaaP, 2022-02-12:
Current implementation:

https://user-images.githubusercontent.com/21154023/153705083-3e4b4f3d-71ec-4850-8e75-ff3a94dd8594.mp4

Official implementation:

https://user-images.githubusercontent.com/21154023/153705216-45ea1705-75c3-4ac6-a25e-4266b59736de.mp4

Official driver does a smooth transition between pressed & non pressed state, making the camera smoother in most games.

Ryochan7, 2022-02-13:
The Steam Input implementation has somewhat aggressive smoothing enabled by default. SC Controller doesn't add smoothing to that output type by default IIRC. Adding smoothing and playing with the parameters would probably help a fair amount.

Not entirely sure how accurate the final output would be from the output though. I mostly map the Steam Controller to a virtual KB+M config.

mochaaP, 2022-02-15:
I noticed that the "Smooth" checkbox in the configuration interface is disabled for Joystick Camera. Not sure if it's intended or it's only for mouse output?

Ryochan7, 2022-04-17:
SC Controller actively removes support for Smoothing when the XY action points to AxisAction instances which would be used in this case. AxisAction does not allow smoothing to be used for some reason. Might have to look into changing that.

Ryochan7, 2023-03-08:
Looks like I will not get around to changing it

closed 2023-03-08 by Ryochan7