Ryochan7/sc-controller #56: Unable to get the triggers to be read as fully pressed without having to click them

hazelnot, 2022-03-24:
I can't get the triggers to count as fully pressed without clicking them, which means I can't really use this thing for racing games without my index fingers hurting the whole time from having to apply a lot of pressure for the accelerator.

Not even calibrating it with jstest made it work, it read the axis as fully pressed but in-game I still had to click it

Ryochan7, 2022-03-24:
I believe that is the intended behavior with the Steam Controller for the most part. IIRC, the Steam Controller sends a dual stage trigger full press click as a button flag in the HID input report and SC Controller checks that flag to engage the full press action. When looking at the raw values coming from the controller, it is very hard to have the trigger hit the max input value without engaging that trigger click button.

Analog emulation mode could probably be changed to add a max zone to the trigger input so the output value will max out before engaging the trigger click. That should work for your use case and that shouldn't be too difficult to add in.

hazelnot, 2022-03-24:
That would be amazing!

Ryochan7, 2022-04-02:
Got around to it. Commit 6ab29a9b562eb5e48c3a5ccb068bd98d1309183a should be good enough to allow that mapping on the triggers. Ended up having to change some calculations for the DeadZone modifier as well.

Never mind. Looks like editing the output range of the analog output is still not possible. I'll have to make more edits later.

[image]

Ryochan7, 2022-04-03:
Some weird quirks involved but I got it figured out. 886a3136fafbb4d00894b8ef11f5324a71d80309 now handles the use case for sure.

closed 2022-04-07 by Ryochan7

hazelnot, 2022-04-08:
Oh wow thanks a lot! I'm making a video about a racing game and my Xbox controller broke which was what prompted me to use the Steam Controller and run into this haha