Ryochan7/sc-controller #57: scc-daemon crash when using blootooth paired steam controller

labsin, 2022-04-17:
I get the following error when using a bluetooth paired steam controller:
```
$ scc-daemon debug
D SCCDaemon     Starting SCCDaemon...
D SCCDaemon     Initializing drivers...
W SCCDaemon     Skipping disabled driver 'fake'
W SCCDaemon     Skipping disabled driver 'remotepad'
D USB           Registered USB driver for 054c:09cc
D USB           Registered USB driver for 28de:1102
D USB           Registered USB driver for 28de:1142
D Mapper        Creating virtual devices
D Mapper        Keyboard: <scc.uinput.Keyboard object at 0x7ff26bd26bc0>
D Mapper        Mouse:    <scc.uinput.Mouse object at 0x7ff26bd26a10>
D Mapper        Gamepad:  <scc.uinput.UInput object at 0x7ff26bd26500>
D SCCDaemon     Created control socket /home/sams/.config/scc/daemon.socket
D SCCDaemon     Connected to XServer :0
D asyncio       Using selector: EpollSelector
D OSD DaemonCtrl    Connected to daemon, version 0.4.8.6
D OSD DaemonCtrl    Daemon is ready.
I SCCDaemon     Registered scc-osd-daemon
I OSD osd.daemon    Sucessfully registered as scc-osd-daemon
E SCBT          [Errno 5] Input/output error
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/scc/drivers/sc_by_bt.py", line 110, in new_device_callback
    return SCByBt(self, syspath, dev)
  File "/usr/lib/python3.10/site-packages/scc/drivers/sc_by_bt.py", line 141, in __init__
    self.flush()
  File "/usr/lib/python3.10/site-packages/scc/drivers/sc_by_bt.py", line 247, in flush
    self._hidrawdev.sendFeatureReport(msg)
  File "/usr/lib/python3.10/site-packages/scc/lib/hidraw.py", line 117, in sendFeatureReport
    self._ioctl(
  File "/usr/lib/python3.10/site-packages/scc/lib/hidraw.py", line 59, in _ioctl
    result = fcntl.ioctl(self._device, func, arg, mutate_flag)
OSError: [Errno 5] Input/output error

```
Any idea what I could try?
System:
Manjaro linux
Kernel 5.10
It seems the steam controller is connected with the linux build in module. Could that be it?

Ryochan7, 2022-04-17:
I'll have to look at that later. It has been about a couple of months since I last tested Bluetooth support but it worked at that time. I usually use the Steam Controller wired but sometimes I use the SC dongle when I connect the controller wirelessly.

labsin, 2022-04-18:
I confirm it works fine with the SC dongle.
The dongle is not so bad as a workaround.
Please let me know if I can help test it or you need more info.

Ryochan7, 2022-04-18:
Tried it out on Fedora 35 and sending the initial feature report fails so no mode switch occurs. I tried the same routine on my older Fedora 33 install and SC Controller can interface with a BT connected Steam Controller just fine. At least I know the problem doesn't lie with the project code. Not sure what the deal is though.

Ryochan7, 2022-04-19:
I am having no luck figuring this out. Tried in the latest Manjaro and it fails for me as well. However, BLE support still works fine in the latest version of Pop!_OS (21.10). So far, the only common factor that might matter is the version of BlueZ that a distro uses. Both Fedora 33 and Pop!_OS 21.10 use BlueZ 5.60. Fedora 35 and the latest Manjaro use BlueZ 5.64. Older updates to BlueZ (around 5.56) did break BLE support in the Steam client at one point so that seems like it could be a reason why support could be broken.

https://github.com/ValveSoftware/steam-for-linux/issues/7697

It is too bad I cannot really find any info or code outside of SC Controller on how to interface with the Steam Controller via Bluetooth.

Ryochan7, 2022-04-19:
Fedora 34 gave a more concrete test of my theory. Fedora 34 comes with BlueZ 5.56 by default and BLE support works fine then. BlueZ 5.64 is available in the updates repo and BLE support breaks once that version is installed. Reverting back to version 5.56 allows the controller to work again. So something is happening with recent updates to BlueZ that does not allow feature reports to work as expected.

Ryochan7, 2022-04-19:
BlueZ 5.61 is also confirmed to work. It is the oldest BlueZ package available in the Fedora 35 repos.

Ryochan7, 2022-04-19:
It doesn't make sense given the available documentation but I might have stumbled across something. Finally got some data sent as a feature report to the Steam Controller when using BlueZ 5.64. Just made a small change in the SC Controller code and the app can actually interact with the controller again. Going to have to play with it some more just to make sure it is not a fluke. I have already double checked that I am currently using BlueZ 5.64.

Based on testing, data has to be sent with feature report ID 3 as the first byte of the feature report. Otherwise the Input/output error occurs.

labsin, 2022-04-19:
Think I found it:
https://github.com/bluez/bluez/commit/35a2c50437cca4d26ac6537ce3a964bb509c9b62
https://github.com/bluez/bluez/issues/15

Valve also commented on the commit

Ryochan7, 2022-04-20:
Got to confirm the change would not affect the behavior with older versions of BlueZ. BLE support works with the small change in commit 321e6de4aa1994228082a30dbc9161e36a3d38f2.

labsin, 2022-04-20:
:ballot_box_with_check: Confirm that it's fixed like this

closed 2022-04-20 by labsin