Ryochan7/sc-controller #58: Steam controller over BLE only connecting on startup

labsin, 2022-04-20:
To reproduce:
* Start daemon
* Power on Steam Controller for pairing over BLE
* Steam controller not recognized by SC Controller
How it does work:
* Start controller
* Start daemon
* Controller recognized

If I disable and enable emulation in the GUI after turning on the controller, it is also recognized.

Ryochan7, 2022-04-21:
The problem stemmed from a couple of different issues. One was a callback signature problem and the other was just polling for data too early while attempting to verify a newly connected device. Hotplug detection for BT connected Steam Controllers has been fixed in commit 43d148327a1b92042d67dd6cfdf1128aa6f9b25d.

labsin, 2022-04-22:
That was quick.
Confirmed that it's working well now :+1: