Ryochan7/sc-controller #59: Steam Controller over BLE only turns off after another input

labsin, 2022-04-20:
To reproduce:
* Press 'turn off controller' from the start menu
-> Controller doesn't turn off
* Press any button
-> Controller turns off

If I bind start + Y to turn off the controller like it is standard in steam, you don't need a second input.
Works fine when using the dongle.

Ryochan7, 2022-04-21:
I would assume that the routine probably waits until the end of the next gamepad poll before the disconnect occurs. Dealing with the input reports in BLE is pretty funky compared to wired or with the SC dongle. If the mapper is responsible for handling the event with new data then there would likely have to be some physical input change to make sure the mapper input routine runs; I would think gyro input would be enough as long as the controller is not resting on a surface.

I will look into it.

labsin, 2022-04-22:
Adding a self._driver.flush() here, fixes it:
https://github.com/Ryochan7/sc-controller/blob/868e341402dc18d40d5d3fd5d8d13feed8e69f14/scc/drivers/sc_dongle.py#L391

Ryochan7, 2022-05-05:
Finally got around to looking into this. A flush call is required to have scc actually send the packet to the controller for a BT connection. Without that, scc waits until the next call to the _input method is invoked.

I made the change in commit b5e0572f52119f39b676e2cca1b4f6f1d0bd327d. Although, I chose to override the turnoff method in sc_by_bt.py rather than edit the base SCController turnoff method.

Ryochan7, 2022-05-06:
Got a new release published with the fix. Gave you credit in the release notes. It is probably safe to close this issue now. Thank you for looking into the various BLE support problems.

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.7

closed 2022-05-06 by Ryochan7