Ryochan7/sc-controller #60: Potential incomplete data in BLE reader

Ryochan7, 2022-04-21:
People won't like it but I have been working on getting BLE support working on my Windows Steam Controller mapper; AFAIK, no other Steam Controller mapper on Windows has support for it outside of Steam Input. Anyway, I had noticed packet state problems while reading reports that would cause Action Layer changes to activate unintentionally. Tweaking the init sequence and HID device reader only got so far. Then I noticed an issue with the current assumption regarding input packet states that might not be true that would also affect SC Controller.

Looking at the current SC Controller code and the data offset index seemed to indicate that the current routine could go outside the bounds of the true data given provided that all elements (Buttons, Pads, Gyro, etc.) are in use at the same time; luckily a 256 byte array is used in the reader so there is no way the data offset will go outside the bounds of the array. Testing it out in SC Controller seems to prove that the current way long packets are dealt with is wrong in a couple of ways. The one case I care about now is that it looks like the Steam Controller can send a payload as 3 packets rather than up to 2 packets like the current BLE reader assumes.

```
Broken sequence: 0x80 at byte 1 denotes long packet flag. SC Controller assumes that the Steam Controller will send
payload data in at most two input packets. That assumption seems to be wrong. 0xC2 seems to denote a third packet to
read in an input payload data

03 80 94 1f 0b 01 1e be f8 ff 7f 6a 0a b6 38 ca fc b0 14 ae 81 f0 7e 20 7c 35 b7 00 67 00 4b 00 e0 29 f8 1e b7 e7 a2 00

03 c2 8d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

```
Semi-normal sequence: 0xC0 at byte 1 denotes single packet data. 0x80 is the long packet flag

03 80 04 1c 63 01 86 fd 29 3f 00 00 00 00 00 00 f5 7f 89 fd c1 5c fe 18 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

03 c0 04 0c 94 01 85 fd 1b 3f ff ff 01 00 00 00 00 00 00 00

03 80 04 1c ac 01 a6 fd f2 3e 02 00 01 00 00 00 f5 7f 89 fd c1 5b fe 18 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

03 c0 04 0c c8 01 92 fd fa 3e 01 00 01 00 00 00 00 00 00 00

03 80 04 1c 7d 01 75 fd f8 3e 01 00 00 00 01 00 f5 7f 8a fd c1 5b fe 18 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

03 c0 04 0c a4 01 81 fd f8 3e 03 00 ff ff 01 00 00 00 00 00
```

This could also be contributing to the problem mentioned in issue #59. This issue is partially my fault as the payload would have been bound within 2 packets when using the old gyro flags in the configure sequence. The flags changed so the accel data would be pulled from the controller for use in CemuHook.

Ryochan7, 2022-04-22:
I believe I found a decent way to take longer input payloads into account with commit 6d48230c097db1d1a7e439d414435a997ce0600b. Going to test it more thoroughly later while playing the final chapter of Remember Me.

Ryochan7, 2022-04-22:
Made a couple more fixes in some new commits. At least the Windows version is doing what I want now. Need to load up Saints Row 4 to test these changes properly in SC Controller.

Ryochan7, 2022-04-22:
Seems to be working as intended in SC Controller as well. Going to have to play around with it some more before I can consider closing this issue.

sagaopc, 2022-04-25:
Any chance to port this to windows?

Ryochan7, 2022-04-25:
If you mean updating the existing C port of SC Controller for Windows then the answer is no. Right now I am testing the waters with my own mapper currently written in C#. There is no real GUI outside of one button telling the app to read a JSON profile and start processing input. The original target device for the test mapper was the Switch Pro controller but I have been experimenting with the Steam Controller for the past few weeks.

Already played through some games with it. Still messing around with the routines to work out some more use cases. Getting stackable Action Layer support working really opened up some mapping options and led to more active development lately. Not sure if I will get it to the point where it will be more user oriented.

Remember Me - Steam Controller mapper test
https://www.youtube.com/watch?v=-QvJoQMGupg

Steam Controller test repo:
https://gitlab.com/ryochan7/steam-controller-test/-/tree/action_ideas

Original Switch Pro test repo:
https://gitlab.com/ryochan7/switchpromappertest/-/tree/action_ideas

closed 2022-05-06 by Ryochan7