Ryochan7/sc-controller #61: Need calibration for controllers with gyroscope.

arkdae, 2022-04-23:
The first thing I did after installing SC-controller is I tried to set up the gyroscope on my Dual Shock 4 to send mouse events. As soon as I did so, the mouse started creeping slowly to the upper left corner. Please note, that my controller is sitting still on a very sturdy desk. It should not creep.

In Steam this can happen, and either Steam will figure it out (I assume, as it eventually stops), or I can run the calibration routine, and it stops.

This utility needs that functionality, as it would be impossible to make use of the gyroscope effectively in games if your view is always moving and you are always compensating for it.

Ryochan7, 2022-04-23:
I cannot remember for sure but I believe that SC Controller does not enforce a default dead zone for gyro mouse. You have to use the dead zone modifier but I am not sure how it would work out with gyro coordinates input; I had to change that modifier recently to work more correctly for trigger input. That would be the first item to worry about.

SC Controller could use some form of gyro calibration routine. Using a simple average would probably be good enough but it might be better to port the JoyShockMapper calibration routine. That routine was ported to DS4Windows a while back and it really helped when using other controllers in my experience; didn't seem to affect the DS4 experience much for me.

I have tried Gyro with the Steam Controller in both Steam Input and SC Controller several times and never liked it. I thought maybe the IMU in the Steam Controller just wasn't up to the task. Only recently have I made a configuration with my little test mapper that I found Gyro Mouse to be usable with the Steam Controller. I have played through many games using the DS4 and Gyro Mouse though.

arkdae, 2022-04-24:
I took at look at JoyShockMapper, and I see that there is a C++ header file for handling the Gyroscope and accelerometer calibration here:
https://github.com/JibbSmart/GamepadMotionHelpers

Doesn't look too hard to make a Python wrapper around it.

arkdae, 2022-04-25:
I was able to wrap it and generate a Python module using Pybind11. Next I need to figure out where to hook it in.

IDeathByte, 2022-05-10:
Hm. Maybe my idea are wrong and i miss some things, but how about add "human factor" filter for prevent drift?

As i see most times, when drift starts, it has linear character - constant speed to move on x-y axes.
Humans, as i think, cant support constant acceleration on their body moves - it will variable time by time and change their direction (jitter). 

Idea is catch data, whats has fully linear character, and subtract it from gyro movement data.

arkdae, 2022-05-13:
That may be useful, but the first thing is that the controller's gyroscope and accelerometer should not cause the mouse pointer to drift while it is just sitting on a stable desk.

I found the code in JoyShock mapper which takes both gyro and accelerometer to help it determine what the controller is doing called "fused input" if I remember correctly. And you can tell it to auto calibrate while sitting down. It'll grab some samples from the controller, and then when done will know how much change in those inputs to ignore.

Ryochan7, 2023-07-26:
Hopefully I will get around to making a Python implementation of the Gyro averaging calibration routine. Only recently have I gotten around to extracting the C# code from DS4Windows into my test mappers.