Ryochan7/sc-controller #64: [Feature Request] Touching pad displays, and activates menu options until pad is released.

ryancordin, 2022-06-02:
In the software, the two main options that I see for activating menu items with touchpads are by pressing the touchpads down, or by releasing them. The part that is highlighted last is the action that fires when either of these two methods are used. The feature request i'd like to make is to include a third option for activating a menu item, and that is by touch activation (highlighted menu item fires immediately, without needing a press or a release).

The equivalent options in Steam can be found in Radial Menu/Touch Menu, under the section labeled "Menu Activation Style." The options listed there are Button Click, Button Release, Touch Release/Modeshift End, and Always. 'Button Click' and 'Touch Release' behave similar to what can be done currently. The section called 'Always' is what I described. It's incredibly useful for selecting menu items quickly, or for activating menu items continuously (repeat keys, etc.). In addition, it frees the touchpad button so it could be assigned to some other function, such as activating an alternate menu (Modeshift).

Ryochan7, 2023-07-26:
Just now seeing this after over a year. I am not familiar with that touchpad option in Steam Input. I guess that would have some valid use cases. I might get around to seeing if such an option can be easy to add to this project.

ProtonAbyss, 2023-07-27:
Thanks for considering the feature.