Ryochan7/sc-controller #65: [Dualshock 4] Triggers not being remapped as analog

vagner995, 2022-06-03:
Hi. I'm using sc-controller-git in manjaro through the AUR repository sc-controller-git. It's using version 0.4.8.7 of sc-controller, which I think is Ryochan7's fork. I've been using it with my dualshock 4 and it has mostly worked.

My DS4.v2's inputs all work fine with just the driver that comes with the kernel. Every analog axis is reading correctly without trying to emulate a xbox controller. The problem is: when emulating the controller, no matter what I do, the triggers are not analog. They only remap to 3 values: -32767, 258 and 32767 in a fashion of <50%, exactly 50% and >50%. This appears reading the KDE settings controller tab. Is this a bug with dualshock 4's or a limitation in the emulation of the controller? I don't know how else to demonstrate this but to record the screen, but I think it should be easily reproduced by anyone with a DS4.

Ryochan7, 2022-06-06:
I'll have to look into that in a bit. SC Controller can emulate the full range of an Xbox 360 output trigger so any problem would likely be related to the ds4drv input module being used in SC Controller.

Ryochan7, 2022-06-07:
Can you share the profile you are using? Trigger mapping works fine for me using a DS4 v.2 and DS4 v.1.

vagner995, 2022-06-07:
Sure. It's just the default. Only change I made was to switch between the profiles with a click of the ps button. I tried making some combinations work to shut off the controller too, but none worked so I settled with just holding the ps button 10 seconds. Anything else you need, just ask.

[XBox Controller - Default.txt](https://github.com/Ryochan7/sc-controller/files/8849832/XBox.Controller.-.Default.txt)
.

Ryochan7, 2022-06-11:
Nothing is wrong with the profile. Not sure what the problem is here. I guess I should mention that I have SC Controller using the ds4drv module with the HID backend enabled. I don't really support the use of the evdev backend module when it comes to DS4 support as the `HID device support` option is superior.

Not too worried about the lack of LED lightbar options with the DS4. There is no real brightness setting for the DS4 lightbar in the DS4 HID output report so any option given would have to fake it with darker versions of a set color. The current LED brightness controller setting still works with the Steam Controller.

vagner995, 2022-06-11:
Just noticed that without "Enable evdev support" my controller isn't even detected by scc. For some reason the HID backend isn't working with scc, but works with ds4drv just like the trigger remapping. Would it help sending whatever config file is used by the ds4 for input? If so, where would I find it?

closed 2022-10-06 by Ryochan7