Ryochan7/sc-controller #66: Steam Controller Left Thumbstick Press

cditi89, 2022-06-09:
Hello,

There seems to be an issue with the left thumb stick press that has developed recently in my Arch Linux setup. I installed SC-Controller from AUR. Everything is updated to the latest on the system. I'm using KDE and Xorg. All dependencies are satisfied. I'm using the Steam controller with the included dongle wirelessly. I want to see if it's just a "me" issue or others are seeing it as well.

SC-Daemon sees my Steam controller plugged in and on any other button press it reports a press, but the left thumbstick click. If I stop SC-controller emulation and open Steam so that my Steam controller is recognized by Steam, the left thumbstick click works bringing up the Steam keyboard.

I have tested SC-Controller with a PS4 controller and the click works fine. Is there anything else to check SC-Controller-wise? Could there be an issue with my current settings somewhere?

Ryochan7, 2022-06-11:
Just tested this using a normal USB connection and using the SC Dongle and Left Thumb Press works fine for me. Not sure what the problem would be.

cditi89, 2022-06-11:
It magically fixed itself on my end? Don't know what happened.

closed 2023-03-08 by Ryochan7

Ryochan7, 2023-03-08:
Not sure if it would be related but maybe it has something to do with how SC Controller attempts to resolve conflicts between Left Touchpad coordinates and Thumbstick coordinates; I cannot remember how or if it attempts to resolve it. Both inputs use the same bytes for both coordinates in the input report and will just alternate between packets. Don't know why Valve did that. There is a flag in the input report that specifies if both are enabled at the same time though so that can be used to help resolve which input is being used in an input report packet. The previous state has to be saved so the previous coordinates can be transferred.