Ryochan7/sc-controller #67: Can this be used on the steam deck?

syberphunk, 2022-06-20:
I've heard of people using this in linux to allow for the steam controller to work with stadia in google chrome, so I'm wondering if this could come to the SteamOS for the steam deck?

So that the 'usb connected steam controller' that is the steam deck could have better control/functionality, and also work with stadia. Though SteamOS mainly installs stuff through https://flatpak.org/ (I'm aware there's an aur package https://aur.archlinux.org/packages/sc-controller-git ) but it'd be better as a flatpak permission wise.

Ryochan7, 2022-06-21:
I don't know much about SteamOS 3.0. I would assume that getting this fork of SC Controller working in Desktop mode (or whatever it is called now) would be possible. Not sure how useful it would be overall especially without Steam Deck controller support. There was some discussion on the Kozec Python 2 version repo with people trying to decode the packet payload for the Deck controller but I don't think anyone has come up with a full breakdown as of yet.

As for Flatpak, I have barely ever worked on packaging apps for it. I think I packaged my old Dreamcast Now QML app for it but I cannot remember how I did it.

This probably won't happen tbh.

syberphunk, 2022-06-21:
I was able to compile it by following:

> passwd
> sudo steamos-readonly disable
> sudo pacman-key --init
> sudo pacman-key --populate archlinux
> wget https://aur.archlinux.org/cgit/aur.git/snapshot/sc-controller-git.tar.gz
> tar -xvf sc-controller-git.tar.gz
> cd sc-controller-git
> sudo pacman -S python-setuptools python-cairo python-pylibacl xorg-xinput glibc linux-api-headers zlib
> makepkg
> pacman -U sc-controller-git-v0.4.8.7.r0.929b3c4c-1-any.pkg.tar.zst

And then I could run it in desktop mode. The main reason why I was investigating this, is because the steam deck's controller isn't identified by Google Stadia as a controller in Google Chrome, and I found people had used sc-controller as a solution to that.

The steam deck's controller identifies as a "steam controller connected via USB" but obviously its functionally different.

It is also possible to connect steam controllers to the steam deck, either over USB, dongle or bluetooth though.

cyrozap, 2022-08-19:
@Ryochan7 As of kozec/sc-controller@4d0e6c7d469bc14119adfbf90ea2bcc45e331b95, the Python 2 version has support for the Steam Deck's controller. Would you be willing to port those changes to your Python 3 version?

bell07, 2022-09-29:
@Ryochan7 , The SteamDeck support is important for Gentoo users, since the Python 2 is not in Gentoo portage anymore. I got the kozec fork working on my SteamDeck, using local Python-2 and pip installation. I am very interested to use your fork on SteamDeck and cleanup all the manual installations, since your fork is used in Gentoo Portage.

Ryochan7, 2022-09-30:
I haven't kept track of the upstream Kozec repo so I did not realize that Steam Deck controller support was incorporated in a main branch. I can look into porting over the changes a bit later. Obviously, I cannot test it since I do not have a Steam Deck.

bell07, 2022-10-08:
Tried the git version python3 branch using live-ebuild (based on portage's one).

Basically it does work on steam deck now, but I have issues.

The OSD menu is hanging if I try to go into "Switch to" or "Autoswitch Options" submenu.

I started with empty `~/config/scc` and run the the tool using 'sc-controller' from command line. Do not see any error messages, but need to kill scc-osd-daemon.py. How I can debug the issue?

Ryochan7, 2022-10-08:
Have not had a chance to really use the app. Just made sure the app launched, could detect a Steam Controller, and also changed some app settings. Spent time yesterday tweaking the appimage-build script to build an AppImage that would work properly on Ubuntu and PopOS; the test AppImage has also been tested on Fedora 36 and the latest Manjaro.

Testing out the Autoswitch Options now and the app does hang for me as well. Running the daemon in debug mode shows that a TypeError exception occurs when trying to retrieve an x window property.

Ryochan7, 2022-10-08:
Autoswitch Options has been fixed with commit 8451760bca5f143734ba6d8ddca1b3fb29159404

bell07, 2022-10-09:
Thanks. Confirm the Autoswitch and "Autoswitch Options" OSD submenu is working.
The "Switch To" submenu is still broken.
Tried another OSD features. All others seems working.

Ryochan7, 2022-10-09:
Pretty much the same problem. Got it fixed.

bell07, 2022-10-09:
Thank you! Now this issue is solved for me. The 0.4.8.9 is fully useable on my Steam Deck with Gentoo  for now!

cyrozap, 2022-10-14:
This works for me, too, now, on Arch Linux on a Steam Deck. It doesn't really work in Wayland (OSD and systray icon don't appear), but that's a separate issue, and it otherwise works fine with X.

closed 2023-04-15 by Ryochan7