Ryochan7/sc-controller #70: [Feature request] Dualsense output

IDeathByte, 2022-07-22:
It possible to realize? 
As i know, in supported titles, dualsense/shock support their features like touch panel/haptic events (mostly in wired mode) out of box.
Idea is translate events from sony to steam controller/deck api for realize many native functions, like reactions for surfaces, events, actions with touchpanels (through layer shifts) etc.

DominikNovosel, 2022-10-15:
https://github.com/Ryochan7/sc-controller/issues/10#issuecomment-750911949

DominikNovosel, 2022-10-15:
Also take a look at this branch: https://github.com/Ryochan7/sc-controller/tree/theY4Kman-dualsense

IDeathByte, 2022-10-15:
I mean not exactly this. 
Idea is make possible for gamepads with haptic feedback(steam/nintendo/8bdo etc.) can emulate part of dualsense features by create virtual controller and recieve haptics commands. For now we has only virtual x360 emulation option.

DominikNovosel, 2022-10-15:
That's a good idea, but the problem is that neither the Dualsense nor the Steam Deck's hardware have been sufficiently documented to enable them to fully work with third-party software.

IDeathByte, 2022-10-15:
How about workaround? Recently, i seen this project https://github.com/Mutcholoko/Haptic-Feedback-Linux
As i understand principle, we make udev rule for create new sound device in dualsense connection, and it just speaker.
Not sure how it works inside gamepad, but as input we receive some sound data.
As i see, in sc-controller we can regulate force and freq of haptics feedback, what's in theory getting us possibility to separation feedback data for dualsense, and with some function layer, translate this data to features, what's we has now (also we can try to look at this project with singing controller https://gitlab.com/Pilatomic/SteamControllerSinger , maybe it has something usable for this )
Also i seem some projects, whats getting us to manualy configure haptic feedback of dualsense, but without receiving data https://github.com/Paliverse/DualSenseX , what's we also can use in theory for conversion.
And in the raw end, we only haven't data, whats dualsense receive from game(maybe it just wav heh).
Idk how it really works inside and how hard to realize that, but i think if it possible, it will be bomb.
Thx for your time :)

FZXizzbKv7jg, 2022-11-14:
Work together with yoyossef. Try to convince him. I think sc-controller > ds360

closed 2023-07-26 by Ryochan7