Ryochan7/sc-controller #71: 8BitDo wireless adapter. add multiple controllers.

Vosester, 2022-09-04:
Hi 
When using 8BitDo wireless adapter(2dc8:3107 8BitDo IDLE).
It creates 50+ controllers that cause instability.
[image]

Output from terminal in -d mode spams.
```
Daemon reported profile change for ev2CBF26E1: /home/user/.config/scc/profiles/Ahat  in time.sccprofile
Daemon reported unknown profile: 'None'; Overriding.
```
The ev2CBF26E1 ID changes but the error its the same.

The issue also appears in the original python2 build.
[](url)https://github.com/kozec/sc-controller/issues/611

I have tried both on python 2&3 builds on arch, my best theory is that the way 8BitDo programed the adapter for compatibility with PlayStation,Xbox, Switch, etc.
Is causing quirkiness.
Four JS controllers are created by evdev and each one in turn is registering as multiple controller types by sc-controller , this causes a recursive adding of many controllers each getting a profile. 
The controller works but sc-controller becomes unstable and locks up.

On a direct connection via Bluetooth the controller works, but also has issues.
One is that sc-controller registers buttons as multiple press e.g L1 and R1 both  light up when ether of them are pressed, even after manual biding. 

The other issue not to do with sc-controller, is major lag and instability when used with other Bluetooth devices on the same Bluetooth controller. 
The reason I purchased the adapter was to solve that issue. 

The best way I was thinking of for a fix, is a toggle to ignore all controller types that are not Xbox. 
I will dig into the code when I get time.

Vosester, 2022-09-05:
After a quick look at the code I found this.

sc-controller/scc/drivers/evdevdrv.py/

```
def _generate_id(self):
		"""
		ID is generated as 'ev' + upper_case(hex(crc32(device name + X)))
		where 'X' starts as 0 and increases as controllers with same name are
		connected.
		"""
		magic_number = 0
		id = None
		while id is None or id in self.daemon.get_active_ids():
			crc32 = binascii.crc32(b"%s%d" % (bytes(self.device.name, 'utf-8'), magic_number))
			id = "ev%s" % (hex(crc32).upper().strip("-0X"),)
			magic_number += 1
		return id

```
This looks like where SC-Controller is grabbing all the evdev controllers and giving them an ID, this explains the ev***** spam.

toby63, 2022-10-11:
I also have this problem from time to time.
Suddenly so many controllers appear that the system is barely usable or I even have to make a forced reboot.