Ryochan7/sc-controller #74: Mouse pointer does not move when analog stick is not moving

chewi, 2022-11-09:
In short, this is a repeat of kozec/sc-controller/issues/576, which was fixed in kozec/sc-controller@2cddbc2eee7e9066621870f381326ac4062519ce, but the code was in C back then, and has since been rewritten in Python. kozec/sc-controller/issues/668 includes a good demonstration of the issue. Simply selecting the built-in Desktop profile is enough to see it. You really notice the difference when you wiggle the stick versus when you just hold it in one place.

chewi, 2022-11-09:
Actually I've now realised this issue is not quite the same. In this case, the pointer jumps after moving the stick. In the original issue, it continued moving from where it was before. I had a go at implementing something similar to what was done before, but it didn't quite work right, and this is probably why.

Ryochan7, 2023-03-08:
Got a preliminary draft going. Making the change also required changing the mapper class and the Mouse uinput class. Scaling is done in the Mouse class that would really throttle any mouse output from an analog stick. Currently have a flag in place to skip any extra scaling in that layer. That way the MouseAction class will be responsible for any applicable scaling.

Also, there seems to be an implicit dead zone even when one is not assigned in the profile. The dead zone is axial which means that proper circular movement cannot be achieved.

chewi, 2023-03-09:
Has it been 5 months already? :laughing: Sounds like a good start, thanks for your efforts.

Ryochan7, 2023-04-15:
This problem is resolved at least when it comes to RS to mouse translation. Tested on the DualSense. Now RS Mouse works more like what I am used to from my other experiments.

https://github.com/Ryochan7/sc-controller/commit/fa32bb3b570eadeebef9f96bc3027de3ddd0e56d

chewi, 2023-04-16:
I've given it a brief try. It's certainly much better now, so thanks! It's still a little jumpy, but maybe it's difficult to fix properly. I've since finished and uninstalled the game I wanted this for, but I expect I'll need it for another one at some point.