Ryochan7/sc-controller #75: Right analog stick does not work as mouse

bell07, 2022-11-10:
Configured the right stick to use as mouse for old game on Steam Deck with Gentoo. 

```
    "rstick": {
        "action": "mouse()"
    }, 
```

`evtest`on the "SCController Mouse" device returns no mouse input. The mouse is not moved :-(

How I can help to find the reason?

Ryochan7, 2023-04-15:
I finally changed how the mouse action works with RS. It took some hackery but it feels better to me and it is closer to what I am used to from my other experiments. More of the actual computation is done in the mouse action rather than in the Mouse class (uinput.py).

https://github.com/Ryochan7/sc-controller/commit/fa32bb3b570eadeebef9f96bc3027de3ddd0e56d

bell07, 2023-04-16:
unfortunately the change did not fix my issue. The right stick still not work as mouse

Ryochan7, 2023-11-02:
Not sure why HAS_RSTICK is not set in ControllerFlags for the Deck. Actions should definitely not have to know what device is being mapped if it can be helped.

Ryochan7, 2023-11-06:
Now that I have mapper.py pulled up, I can see why HAS_RSTICK is not set for the Steam Deck module. Both the Right Stick and Right Touchpad use the same mapping id (RIGHT). Such a configuration works fine for the DS4 or DualSense as those controllers only include one touchpad (CPAD). It complicates matters when a device has two sticks and two touchpads like the Steam Deck.

mapper.py uses one workaround to use RSTICK as the mapping id in one portion of code. However, later in mapper.input, the conflict will happen. Some extra code will have to be changed. Also, I need to double check what SC Controller uses in the profiles for the Right Stick id.

bell07, 2023-12-09:
Confirm the "ugly fix" does solve my issue. Thank you!