Ryochan7/sc-controller #79: Autoswitch when WM has no windows

gordon-quad, 2023-01-30:
Cannot make scc to switch to Desktop profile when I close all the windows (e.g. exit the game that was using XBox360 profile), looks like it has no options to match such situation in autoswitcher.

Is that a feature request or am I missing something?

Ryochan7, 2023-01-31:
I do not think that feature exists. I do not utilize the auto switcher so it would not affect my use of the program. DS4Windows does have a way to add a default auto profile if no matched entry is found. That functionality could be useful in SC Controller. Hopefully I will get around to it; been out of the loop lately and mostly just been doing Windows stuff tbh.

gordon-quad, 2023-01-31:
I took a stab at debugging it and it looks like there is a functionality to mach root window, but it is broken because of some bytes vs. strings encode/decode business which, I assume, roots from 2to3 migration. Will try to fix and PR it in near future.