Ryochan7/sc-controller #80: Steam Deck stick touch support

gordon-quad, 2023-01-30:
Steam Deck has touch sensors in the sticks' caps, would be nice to see it supported in scc.

Ryochan7, 2023-01-31:
I do not have a Steam Deck so that would be up to someone else to add support for those touch buttons. Been a while since I looked at the source code so I am not sure how much would need to be changed to add two new mappable buttons to the mapper.

Ryochan7, 2023-04-15:
Looking over the steamdeck.py module and there is some placeholder code for handling stick touches. The mapper is fairly open ended when it comes to handling digital buttons. It looks like two new mappable buttons would have to be created for SCButtons (constants.py) to have the daemon work.

Exposing new mappings in the GUI is a bigger challenge. Would need to edit the deck.svg file with a new element and then I am not sure what hook would be required to establish the callback for calling the button edit dialog.

SophiaHadash, 2023-07-22:
I will look into this feature and see if I can add support for the touch events.