Ryochan7/sc-controller #87: Custom Profiles don't load on SC-Controller

gelatinbomb, 2023-05-21:
I did a basic profile to play on my Nintendo 64 emulator and I noticed SC-Controller is not able to load the profile. I can create profiles and delete it without any problem but the moment I try to load it it fails. Loading the program on the terminal shows this each time I try to load the profile:

Traceback (most recent call last):
  File "/tmp/.mount_sc_conS94HFj/usr/lib64/python3.10/site-packages/scc/gui/app.py", line 630, in on_profile_selected
    self.load_profile(giofile)
  File "/tmp/.mount_sc_conS94HFj/usr/lib64/python3.10/site-packages/scc/gui/userdata_manager.py", line 44, in load_profile
    self.on_profile_loaded(profile, giofile)
  File "/tmp/.mount_sc_conS94HFj/usr/lib64/python3.10/site-packages/scc/gui/app.py", line 624, in on_profile_loaded
    b.update()
  File "/tmp/.mount_sc_conS94HFj/usr/lib64/python3.10/site-packages/scc/gui/controller_widget.py", line 105, in update
    txt = self.app.current.buttons[self.id].describe(self.ACTION_CONTEXT)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_sc_conS94HFj/usr/lib64/python3.10/site-packages/scc/actions.py", line 649, in describe
    axis, neg, pos = AxisAction.get_axis_description(self.id)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_sc_conS94HFj/usr/lib64/python3.10/site-packages/scc/actions.py", line 629, in get_axis_description
    if id in Axes or id in Rels:
       ^^^^^^^^^^
  File "/usr/lib64/python3.11/enum.py", line 742, in __contains__
    raise TypeError(
TypeError: unsupported operand type(s) for 'in': 'int' and 'EnumType'

Ryochan7, 2023-05-22:
Ran into this kind of issue with actions before. Supposedly that type of operation was deprecated around Python 3.8.

https://github.com/Ryochan7/sc-controller/commit/c539010f5024781f7529001a2a696f6aa338418f

gelatinbomb, 2023-05-22:
Oh I see. So I should compile from source for now?

Ryochan7, 2023-05-22:
You can but I will try to get a new AppImage out sometime today. It has been a long time since a new release was published and it  needs to be done. I never got back around to tweaking DualSense support further but I guess the current module will have to do.

gelatinbomb, 2023-05-22:
Nice to hear! :D

closed 2023-05-22 by gelatinbomb