Ryochan7/sc-controller #91: On my steam deck, half of the controls stopped working

SophiaHadash, 2023-07-08:
I have a steam deck and a couple months ago controls worked fine. I didn't use it for some time. Now I updated and half of my controls aren't working anymore. Specifically these stopped working:

left thumb axes (press works), right thumb axes (press works), three dots, left trigger, right trigger, dpad, left trackpad move (not click)

The other controls are working fine. I tried to disable the scc-daemon and use the daemon from opensd which resulted in my controls working again. I also tried to start fresh by removing ~/.config/scc but this did not help.

If it is of any help, this is the kernel that I'm using:
Linux deck 6.2.15-zen-00690-g09c4ace4dddd #1 ZEN SMP PREEMPT_DYNAMIC Fri Jun 30 13:25:21 CEST 2023 x86_64 GNU/Linux

Ryochan7, 2023-07-10:
Not sure what the problem could be. I do not have a Steam Deck to test with. My main test targets are the Steam Controller and DualShock 4. Last I checked, the mapper still works fine for both of those controllers. Almost nothing has changed with the Steam Deck module since it was originally added to the project.

You will probably have to try running scc-daemon in debug mode and see what happens. I will try to get around to checking out opensd and see how that application handles input packets from the Steam Deck. I doubt I will be able to really work out the issue without the hardware in hand though. Also, going to put the URL for the project in here for future reference since I had not heard about opensd before.

https://gitlab.com/open-sd/opensd

Ryochan7, 2023-07-10:
Apart from how the first few bytes are handled, it looks like opensd handles the input packet mostly the same as sc-controller.

https://gitlab.com/open-sd/opensd/-/blob/main/src/opensdd/drivers/gamepad/hid_reports.hpp#L103

SophiaHadash, 2023-07-10:
My controls are working on v0.4.8.9 but not on master. I just confirmed.

I can let you use my steam deck if you wish to work on this, just let me know. We can schedule a remote session if you want :)

Ryochan7, 2023-07-11:
Looks like the only major change was this one regarding saving input state. Might have to revert that change but it looks like it should work fine.

https://github.com/Ryochan7/sc-controller/commit/1737c41c49e405c18b91a2e27bcb3d821bdae606

SophiaHadash, 2023-07-11:
I will try tonight on master with this commit reversed.

SophiaHadash, 2023-07-12:
Reverting this commit solves the issue.

SophiaHadash, 2023-07-12:
https://github.com/Ryochan7/sc-controller/pull/92

closed 2023-09-19 by SophiaHadash