Ryochan7/sc-controller #95: Gyro settings randomly cleared

UltraBlackLinux, 2023-08-27:
Hey there,
I haven't been using the gyro a lot, but I was pretty impressed by the Alpakka's performance so I decided to try if I could replicate this with the SC, but I was quite violently stopped by some bug that randomly clears the various different options in the gyro's "Per Axis" configuration menu:
[image]
Whenever I toggle any of those "Absolute", Yaw gets cleared. Whenever I change the sensitivity, Everything gets cleared.
There are a lot more issues with options being randomly cleared or even the whole gyro menu being cleared.
It's hard to miss.
Can you fix this?
Thanks!

Ryochan7, 2023-11-07:
Looks like it is writing the output commands wrong during GUI changes. I now the main Gyro config GUI has had issues as well but those came up so randomly that I never got to find out why. Will look into this issue and see why the GUI is behaving the way it is in the first place.

Ryochan7, 2023-11-07:
I think I found the biggest problem. Had to change an if statement in commit 5c45ca7530c4e96abbd50ac1bb0594653b61537c. ABS_X has a code of 0 which Python will evaluate as false when used in an if statement. Now the routine explicitly checks if the axis value is not None.

At least now that portion of the GUI seems to be working as expected.