Ryochan7/sc-controller #107: Question about "EOL release" tag on latest release - QOL typo?

C0rn3j, 2024-01-21:
I see that there's "EOL release" in the description - is the fork being dropped, or was that just a typoed QOL?

https://github.com/Ryochan7/sc-controller/releases/tag/v0.4.8.13

Ryochan7, 2024-01-21:
I am dropping this fork. I hardly spent any time working on this project in 2023 and I do not see that changing in 2024. Also, I don't really use this software anymore. Video gaming is pretty much a dead hobby to me now.

I have been in the process of making final tweaks and closing several projects I have been working on over the years. Not really getting much out of the time I have spent playing around with FOSS projects. My time would likely be better spent doing other things.

This fork will be archived in a few days.

C0rn3j, 2024-01-21:
That's unfortunate - thank you for maintaining this fork for all these years and an extra thanks for trying to stay in sync with upstream. I will try one more time to appeal having these changes pulled in...

I've been packaging this fork on Arch over the Python 2 upstream and wouldn't mind having it under my account to keep the Issue/PR history and comments intact, would you consider handing the repo ownership over instead of archiving it?